JakobEngel/dso #97: Integrate loop closing / global optimization

deruci, 2017-09-24:
Hi,

I'm looking into integrating features like loop closing / global optimization with DSO.
Although DSO's optimization is done with Information Filter, I think it is possible to integrate these features by constructing another pose-graph.

Is there anyone had tried it?
If it is not that trivial I assume, then why is it?

NikolausDemmel, 2017-10-17:
- You should be able to construct a pose-graph from the keyframes that fall out of the optimization window (similar to LSD-SLAM). With that, the information filter and the SLAM graph are decoupled.
- What is more tricky is integrating DSO into a coupled SLAM pipeline, where you reuse existing keyframes when you revisit that part of the scene. The problem is that it is not straight forward how to deal with the marginalization term in a consistent matter then.

s3r637, 2018-01-04:
Hi @deruci, check this out: https://youtu.be/BxTLhubqEKg I have not yet found a publication but it's just a matter of time.

Cheers

shadowver, 2018-01-19:
@deruci the video looks pretty amazing. I was curious about what framework it used when performing loop closure.

maaft, 2018-03-08:
Any news on publishing the code for "Stereo DSO: Large-Scale Direct Sparse Visual Odometry with Stereo Cameras" including SLAM extension?

NikolausDemmel, 2018-03-08:
No, unfortunately not.

Kallista, 2018-10-22:
I guess this issue is closed with https://github.com/tum-vision/LDSO