JakobEngel/dso #100: dso_dataset: error while loading shared libraries: libboost_system.so.1.58.0

mmf007, 2017-10-13:
Hi!
Please Help me.

I try run dso_dataset:
./dso_dataset files=/DataSet/sequence_14/images.zip  calib=DataSet/sequence_14/camera.txt  gamma=DataSet/sequen$

bin/dso_dataset: error while loading shared libraries: libboost_system.so.1.58.0: cannot open shared object file: No such file or directory

and realy:
ldd dso_dataset
	libboost_system.so.1.58.0 => not found
	libboost_thread.so.1.58.0 => not found
	libzip.so.4 => not found
	libpangolin.so => not found
	libGL.so.1 => not found
	libGLEW.so.1.13 => not found
	libpthread.so.0 => /lib/arm-linux-gnueabi/libpthread.so.0 (0xb6f2a000)
	libopencv_highgui.so.2.4 => not found
	libopencv_core.so.2.4 => not found
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabi/libstdc++.so.6 (0xb6e1b000)
	libm.so.6 => /lib/arm-linux-gnueabi/libm.so.6 (0xb6d85000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabi/libgcc_s.so.1 (0xb6d5c000)
	libc.so.6 => /lib/arm-linux-gnueabi/libc.so.6 (0xb6c6f000)
	/lib/ld-linux.so.3 (0xb6f6f000)

but if call:
ldconfig -p 
....
libboost_system.so.1.58.0 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libboost_system.so.1.58.0
....
libboost_thread.so.1.58.0 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.58.0
....
libzip.so.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libzip.so.4
....
libpangolin.so (libc6,hard-float) => /lib/arm-linux-gnueabi/libpangolin.so
....
libGL.so.1 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/mesa/libGL.so.1
....
libGLEW.so.1.13 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libGLEW.so.1.13
....
libopencv_highgui.so.2.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libopencv_highgui.so.2.4
....
libopencv_core.so.2.4 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libopencv_core.so.2.4
....

My CPU:

lscpu

Architecture:          armv7l
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    1
Core(s) per socket:    4
Socket(s):             2
Model name:            ARMv7 Processor rev 3 (v7l)
CPU max MHz:           1400.0000
CPU min MHz:           200.0000
Hypervisor vendor:     (null)
Virtualization type:   full

My ubuntu:

lsb_release -a

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

Thank you.

NikolausDemmel, 2017-10-17:
Are you running from the same terminal where you compile it? Is there maybe a difference in the `LD_LIBRARY_PATH` environment variable?

SouthEastleo, 2017-10-26:
Hello，I have some questions about the dataset, the camera.txt is created by youself?I have downloaded the mono-dataset but can't find this file.Thanks

NikolausDemmel, 2017-12-07:
@SouthEastleo: The camera.txt should be part of the TUM mono dataset download.

SouthEastleo, 2017-12-08:
Thanks，I have solved the question.thank you a lot.