JakobEngel/dso #102: position output scaled

dbcesar, 2017-10-17:
Hi, first of all, congratulations for the great job. 

I'm using this library on a robot simulation, but when the vehicle moves 1 meter, the output is 0.2m. The same happens in the other axis. 

I'm probably missing something, just don't know what. Camera calibration parameters are right. Do you have any guess? Maybe the output is not in meters? 

Thanks in advance.

NikolausDemmel, 2017-10-17:
The absolute scale is unobservable in DSO. This is an inherent problem of monocular approaches like DSO (at least with geometric techniques). The scale is set more or less arbitrarily during initialization, it will also drift over time (albeit usually slowly).

Therefore, what you observe is expected.

dbcesar, 2017-10-17:
Thank you for the clarification.

NikolausDemmel, 2017-10-17:
Closing this, feel free to continue discussion and we can reopen if needed.

closed 2017-10-17 by NikolausDemmel

bjbraun, 2018-05-26:
Hi dbcesar, how did you output the scale of how far you were moving?

dbcesar, 2018-05-30:
> Hi dbcesar, how did you output the scale of how far you were moving?

Hi, I did not output the scale. I could only see the library output, which did not match with the expected. In order to overcome the out-of-scale output, I performed known movements in order to calibrate the output and apply a scale "corrector".