JakobEngel/dso #103: How to get camera linear and angular speed for every frame ?

taiping-z, 2017-10-21:
Hi, @JakobEngel 

I want to calculate **linear and angular speed** from two consecutive frames. 

Here is what I do.
In the function `void SampleOutputWrapper::publishCamPose(FrameShell* frame, CalibHessian* HCalib)`, after two images input, previous and current camera poses, named `preCamPose, curCamPose` are available. 

Then, I calculate transformation between the previous and current frame.
```
SE3 SE3_pre  = preCamPose.camToWorld; 
SE3 SE3_cur = curCamPose.camToWorld; 
SE3 Transform = SE3_cur*SE3_pre.inverse(); 
```
According to Transform, rotation angle and translation distance can be calculated. 
```
Eigen::Vector3d translation =  Transform.translation();
Eigen::Matrix3d rotation = Transform.rotationMatrix();

double rotate_x = atan2((double)rotation(2,1) , (double)rotation(2,2));
double rotate_y = atan2((double)-rotation(2,0), (double)sqrt((double)(rotation(2,1)*rotation(2,1) +rotation(2,2)*rotation(2,2))));
double rotate_z = atan2((double)rotation(1,0), (double)rotation(0,0));
double trans_x = translation(0);
double trans_y = translation(1);
double trans_z = translation(2);
```
I tested this code based on KITTI dateset.
Here is the **result** when the car goes straight.
```
translation 
0.0652169
-0.0503146
0.00819599
rotationMatrix 
           1  0.000744017 -0.000301069
-0.000743349     0.999997   0.00221291
 0.000302714  -0.00221269     0.999998

rotation x:0.00267967,y:-0.00426949,z:-0.00279313
```
The **result** when the car turns.
```
translation 
-1.07098
-0.000289304
-0.692172
rotationMatrix 
    0.998175 -0.000105276    0.0603865
 0.000103315            1  3.55986e-05
  -0.0603865 -2.92948e-05     0.998175

 rotation x:-0.00172052,y:0.0558483,z:0.000769627
```
**Here is my question**.(Presumed, the x-axis is the forward direction, the z-axis is perpendicular to the ground.) 
When the car goes straight, the translation should be in only one direction x-axis. When the car turns, there should be a large rotation angle around the z-axis. Obviously, this is not the case in my above-tested results.

Am I wrong to calc speed in this way?  What can I do to get translation in one direction, and the rotation angle around the axis perpendicular to the ground?
Any help would be greatly appreciated.

Regards
Taiping
Oct 21, 2017

chabaudmorgan, 2017-10-23:
> (Presumed, the x-axis is the forward direction, the z-axis is perpendicular to the ground.)

There you go, wrong presumption! ^^

Coordinates are in image frame: x-axis is on image columns, y-axis is on image lines, z axis is "forward" (depth)

taiping-z, 2017-10-23:
@chabaudmorgan , thank you very much for your reply. I have already solved this problem.
Regards,
Taiping

NikolausDemmel, 2017-10-23:
Closing, since it seems resolved.

closed 2017-10-23 by NikolausDemmel