JakobEngel/dso #106: ERROR 9 reading archive home/l/Downloads/sequence_14/images.zip!

wowdeath, 2017-11-07:
When I run the DSO with TUM dataest, there is an error occur, it is shown as following:
l-pc:~/DSO/dso-master/build/bin$ ./dso_dataset \files=home/l/Downloads/sequence_14/images.zip \calib=home/l/Downloads/sequence_14/camera.txt \gamma=home/l/Downloads/sequence_14/pcalib.txt \vignette=home/l/Downloads/sequence_14/vignette.png \preset=3 \mode=0 
loading data from home/l/Downloads/sequence_14/images.zip!
loading calibration from home/l/Downloads/sequence_14/camera.txt!
loading gammaCalib from home/l/Downloads/sequence_14/pcalib.txt!
loading vignette from home/l/Downloads/sequence_14/vignette.png!

=============== PRESET Settings: ===============
FAST settings:
- 5x real-time enforcing
- 800 active points
- 4-6 active frames
- 1-4 LM iteration each KF
- 424 x 320 image resolution
==============================================
PHOTOMETRIC MODE WITH CALIBRATION!
ERROR 9 reading archive home/l/Downloads/sequence_14/images.zip!

NikolausDemmel, 2017-11-07:
What happens when you type in `ls home/l/Downloads/sequence_14/images.zip` in the terminal? Is there maybe a leading slash missing?

wowdeath, 2017-11-07:
Thank you very much for your reply. There is no leading slash missing, I have confirm it. But I can't find why the problem happened.

NikolausDemmel, 2017-11-07:
Have you tried with an unzipped dataset?

wowdeath, 2017-11-07:
Yes!Thank you very much that I have solved this problem through your help.
But I meet a new problem, when the camera is moving, the pangolin will exit immediately,and the terminal will show:
Initial Error       	A(513483.839444)=(AV inf). Num: A(0) + M(0); ab 0.000000 0.000000!
STEPS: A 1.4; B 25.4; R 5.9; T 15.6. 	REJECT 0 (L -1.00, dir nan, ss 1.0): 	A(586785.304620)=(AV 7.585). Num: A(1275) + M(0); ab -0.011802 2.155970!
STEPS: A 0.8; B 12.5; R 5.1; T 14.0. 	ACCEPT 1 (L -1.60, dir 0.95, ss 1.0): 	A(529680.450970)=(AV 6.532). Num: A(1552) + M(0); ab -0.018445 3.215516!
STEPS: A 0.2; B 3.2; R 1.4; T 2.6. 	ACCEPT 2 (L -2.20, dir 0.81, ss 1.0): 	A(505022.772114)=(AV 6.409). Num: A(1537) + M(0); ab -0.020195 3.490578!
STEPS: A 0.0; B 0.6; R 0.5; T 1.0. 	ACCEPT 3 (L -2.81, dir -0.04, ss 1.0): A(499328.221278)=(AV 6.402). Num: A(1523) + M(0); ab -0.020598 3.540526!
LOG 8: 6.386 fine. Res: 1523 A, 0 L, 0 M; (0 / 0) forceDrop. a=-0.020598, b=3.540526. Window 8 (2)
Segmentation fault (core dumped).
Thanks a lot.

NikolausDemmel, 2017-11-07:
Hm... yeah this is strange. Same advice as on the other issue. Try to compile in debug mode an check where exactly it crashes.

wowdeath, 2017-11-07:
OK,thank you very much.
And I have an another question. Can you tell me how to compile in debug mode? I'm so sorry that I don't know what is the debug mode.

NikolausDemmel, 2017-11-07:
If you use QtCreator with the cmake project, it usually sets up different configurations for you, so you simply swap to the release configuration. QtCreator also allows you to run with a graphical frontend to the qdb debugger, which is much more convenient than command line. I suggest you read up on QtCreator cmake integration as well as graphical debugging with QtCreator.

To compile as debug build from the command line, you run cmake in your `build` folder `cmake -DCMAKE_BUILD_TYPE=Debug .` and then `make clean` and `make` to rebuild everything.

TarekDakhran, 2018-10-11:
I had the same error when **libzip** was not found or was an incorrect version of **libzip** linked.