JakobEngel/dso #108: Questions regarding Output3DWrapper

m2b821, 2017-11-19:
Hello, I have a few questions regarding Output3DWrapper.
What I try to do: Get point cloud, corresponding frame numbers and scale factor.
My aim is to export the frame number (range), so if there are 7 active frames for each calculation step, how can I gain the corresponding frame ids to them. I have implemented an own Output3DWrapper, using publishKeyframes to find all 3D points until final=true. But I was expecting to get around 2000 points, but I also monitored cases with over 10000 points. Is the approach correct to load all the points from publishKeyframes until final=true?
Second question is about the DSO scale factor, I know that I can get the translation vector for the scale factor from FrameShell. As I have a reference camera path available, I would like to know if this translation vector is from frame to frame or from first frame to last frame (in total 7 frames), as 7 frames are used for each calculation step? Or how is DSO defining the baseline/scale factor for the calculation of the (inverse) depth (I am sorry if I failed to understand this in the DSO paper, but I think that you do normal triangulation at end, or am I wrong?).
Thank you!

NikolausDemmel, 2017-12-07:
I'm not sure what exactly you are trying to do, but if you want to get a pointcloud for all points for each keyframe, maybe just consider each keyframe when it is marginalized (`final == true`) and check all the points. You can decide which ones you want to include (`immaturePoints`, `pointHessians`, `pointHessiansMarginalized`, `pointHessiansOut`). Check how the pangolin viewer does this in `KeyFrameDisplay::setFromKF`. This will also show you that you can access the global keyframe pose as camToWorld.

As explained in other issues, the absolute scale is not observable. The scale of DSO's estimate is fixed somewhat arbitrarily at initialization and then will drift over time. You can compute a relative poses between keyframes using their respective global poses and then update scale if you have some other distance estimate to compare to.

m2b821, 2017-12-07:
I have analyzed the output camToWorld, from which I use the translation only. It looks like DSO is give me the camera path relative to the zero position, so not frame to frame based. I need these information to correct with the simulation camera path for rescaling.

m2b821, 2017-12-18:
> As explained in other issues, the absolute scale is not observable. The scale of DSO's estimate is fixed somewhat arbitrarily at initialization and then will drift over time. You can compute a [...]
I have tried this, assuming I take the key frames and corresponding poses. Then I calculated the difference between the translation of key frame 1 and key frame 2 in case of point cloud corresponding to key frame 2. The absolute movement is known to me from simulation. But the result looks arbitrary. Could you please explain me how DSO is defining the baseline, is this between the exported key frame?

NikolausDemmel, 2018-01-03:
> I have tried this, assuming I take the key frames and corresponding poses. Then I calculated the difference between the translation of key frame 1 and key frame 2 in case of point cloud [...]

Is the localization result (camera path) accurate (except for scale)? You might observe all sorts of strange behaviour, if the odometry does not properly initialize / converge.

Otherwise, getting the scale by computing relative translation between two keyframes should give you ok results. The drift should only be happening slowly over time. If it is changing rapidly, something else is likely going wrong. Keep in mind though, that by simply looking at consecutive keyframes to estimate the scale without any filtering, also small localization errors will translate into scale estimate errors.

m2b821, 2018-01-03:
@NikolausDemmel Thanks, but what do you mean with filtering exactly? Is there any information from DSO how reliable the camera pose is? Besides that, I see a resulting scale factor of around 4, this is rather large compared to SfM, but the increased point number is really nice.

NikolausDemmel, 2018-01-03:
> Thanks, but what do you mean with filtering exactly?

I meant smoothing (e.g. with something like a kalman filter) the scale estimate (estimated translation divided by ground truth translation) as opposed to recomputing it for every frame, based only on the relative pose to the previous keyframe. This is only if you want a scale estimate that is different for every frame, as opposed to one (average) scale estimate for the whole sequence.

> Is there any information from DSO how reliable the camera pose is?

Not that I know of.

> Besides that, I see a resulting scale factor of around 4, this is rather large compared to SfM, but the increased point number is really nice.

Is it roughly constant at 4, changing only slowly. Note that this is somewhat arbitrary. For a different sequence, you might get something completely different. The same should be true for any other monocular SfM technique (I guess you are referring to some traditional feature-based 3d reconstruction). If it's scale estimate is better than DSO's, I believe this should be purely accidental and not a general properly.

m2b821, 2018-01-03:
@NikolausDemmel I want to do it for each keyframe separately, the final ground truth will be hardware sync to the camera output. A global scale factor is the wrong way for me.

NikolausDemmel, 2018-01-03:
> I want to do it for each keyframe separately, the final ground truth will be hardware sync to the camera output. A global scale factor is the wrong way for me.

Makes sense. So are there any open questions here?

closed 2018-01-08 by m2b821