JakobEngel/dso #110: Internal frame identifier - point cloud output strange

m2b821, 2017-11-27:
Hello,
I am using DSO for some simulated images for which I want to get the point cloud and cam poses for each frame. So I set the second parameter in FullSystem to true to enforce each frame as an keyframe (as proposed in issue #51). But when I inspect the fetchted output of my Output3DWrapper using publishKeyframes I see that e.g. frame id 0 returns regularly. I did not get any error regarding initalization or lost tracking. At the moment, I use the internal frame id of DSO. But if the 7 images windows is moved, why do I get frame ids which should correspond to frames from the beginning? Maybe someone could please explain this to me? Beside skipped frames, DSO should be able to deliver a point cloud for each frame after initialization (skipped some frames at the beginning).
I consider only the case finale==false in publishKeyframes, the camera poses for all frames are loading using publishCamPose (results seem to be fine).
My code:
`if(f->shell->poseValid) {
        			currentImage = new Image3D();
					currentImage->frameNr = f->shell->incoming_id;
					for(PointHessian* p : f->pointHessians)
					{
						if (p->idepth_scaled != 0) {
							Point3D* point = new Point3D();
							point->x = p->u;
							point->y = p->v;
							point->z = 1/(p->idepth_scaled);
							currentImage->translation.x = f->shell->camToWorld.translation()[0];
							currentImage->translation.y = f->shell->camToWorld.translation()[1];
							currentImage->translation.z = f->shell->camToWorld.translation()[2];
							currentImage->points.push_back(point);
						}
					}
					images.push_back(currentImage);
        		}`

NikolausDemmel, 2017-12-07:
Why do you consider the case `final == false` and not `final == true` as described in #51? The former will give you all active frame after each BA optimization, while the latter should give you each frame exactly once, just after it has been marginalized.

m2b821, 2017-12-07:
Thank you for your answer.
I am now using final==true, but I when I inspect the results I see the following, after 16 frames, frame with number 0 is keyframe after frame 146. Later also frame 62 is occurring after frame 483. Why is this happening, I did not get any error for tracking lost or so while DSO was running?
Should I suppress these cases or what should I do with them?

NikolausDemmel, 2017-12-07:
The frames are not marginalized sequentially. For example, the oldest keyframe might be kept much longer than some intermediate keyframe. It depends on many things including the motion. So it is perfectly normal that publishKeyframes with `final==true` is called on non sequential order.

m2b821, 2017-12-07:
Ok. Is there a max border how long an old keyframe is staying alive?

m2b821, 2017-12-07:
I have a question regarding the output point number. I use immaturePoints, pointHessians, pointHessiansMarginalized, pointHessiansOut for final==true in publishKeyFrames. When I was seeing Pangolin, I expected that the results have more points. Have I missed something in publishKeyFrames?
Here is the plot of my output from DSO and two example images with DSO points:
[image]
[image]
[image]

m2b821, 2017-12-07:
I don't understand why I see more points in Pangolin than here.

NikolausDemmel, 2017-12-07:
> Ok. Is there a max border how long an old keyframe is staying alive?

I don't think so. If the points stay in view, it might stay active indefinitely, I believe. Check `FullSystem::flagFramesForMarginalization`. You can also see there that if you set `setting_minFrameAge` higher than the number of frames, marginalization should then be sequential it seems. This will probably be a little bit less accurate overall, but could be useful for you.

> I don't understand why I see more points in Pangolin than here.

Not sure what you are seeing in the viewer, but you can see in `KeyFrameDisplay::setFromKF` exactly how the points are extracted. One thing is that Pangolin shows each of the points in the patter, i.e. for one "point" data structure, there are 8 3D points. Is that the difference you are seeing?

m2b821, 2017-12-07:
Ok, multiply by 8 would be valid. Here is a screenshot of Pangolin. I am a bit worried about the low number of points on the cylinder.
[image]

m2b821, 2017-12-07:
If you compare the images from Pangolin with the two images above, there are a lot of points not in my images. But I use with  immaturePoints, pointHessiansMarginalized, pointHessiansOut all the same output I think. But I skipped pointHessians, which returns nothing in case final==true. Might this be the problem that I need to include also final==false cases? In KeyFrameDisplay, I did not find any final decision.

NikolausDemmel, 2017-12-07:
The screenshot you show is not the pangolin viewer (3D), but the "Keyframe debug plot" (see `FullSystem::debugPlotTracking`). I'm not sure what is going on and why you might be missing points on the cylinder. Is that the case for all keyframes?

So the depths in the debug plot (color coding) seems a bit random and all over the place. Does tracking actually work and produce reasonable trajectories?

m2b821, 2017-12-07:
This is not the case for all keyframes, but for some. But if you see the wardrobe in the picture below, there are points missing as if only a part of the pointcloud is returned. I will try to use publishDepthImage and see if this is different. 
[image]
[image]

NikolausDemmel, 2017-12-11:
If you are missing points in some keyframes, you may want to check in the code when points from one of the 4 lists is removed and not at the same time inserted into other lists. This happens for example for immature points in `FullSystem::activatePointsMT`.

What is published as "DepthImage" is always for the latest keyframe the valid point information (points in `pointHessian`) from all other keyframes projected into this latest keyframe and dilated a little bit. This is the depth map that all frames are tracked against. So this is always the current estimate for depth of all active points, and not the final (likely more accurate) estimate. Depending on what you want to do, this might still be more useful for you.

m2b821, 2017-12-16:
Hello, I have found my missing points in some way. But I have a remaining question regarding the camera path: How does DSO calculates the baseline length fro triangulation. Is this always seven frames back (assuming 7 active frames)? Because, I want to rescale it using my simulation camera path information but I did not find the correct setting. Also, DSO own depth information seems very inaccurate for the first 20-40 output frames while getting better at the end. Is this behaviour normal?
See the frames 269 (first frame in DSO), frame 477 and frame 698 below (I have already cut away the largest outliers in terms of length, so about 95% of the 3D points are plotted). For the camera pose: I have always logged it, but I did not find the correct scaling using my defined camera path and the DSO camera path output.
[image]
[image]
[image]

NikolausDemmel, 2018-01-03:
As noted in other issues, the current initializer in DSO is not the best and should be replaced with something more robust.

> Also, DSO own depth information seems very inaccurate for the first 20-40 output frames while getting better at the end. Is this behaviour normal?

If initialization does not work properly, I would say yes. Not sure if that is the issue here, but might be.

One reason why it might not work the best is that large portions of your scene are untextured. I have not tried DSO in such environments and it also wasn't developed for this as far as I know. Might work, might not.

> For the camera pose: I have always logged it, but I did not find the correct scaling using my defined camera path and the DSO camera path output.

Not even manually? If the paths do not align, at all (with some scale factor), that indicates that something is going wrong.

Maybe you can replace the initializer with initialization based on the ground truth from initialization with perfect depth. This is to make sure everything else works as expected. This should also result (initially) in correct scale and only slowly drift away from the correct scale. Once this works, you can revisit initialization from the sensor data without groundtruth.

m2b821, 2018-01-03:
@NikolausDemmel Thank you for your answer. I have no fixed a problem with my scripts and know the results are as good as expected. The first frames remained unreliable, but I will just skip them.

closed 2018-01-03 by m2b821

NikolausDemmel, 2018-01-03:
@m2b821, good to know. Could you maybe share roughly what went wrong for other's reference?

> The first frames remained unreliable, but I will just skip them.

Sounds reasonable to me.

m2b821, 2018-01-03:
@NikolausDemmel: I did a mistake to assume the depth is directly the z value. Now I took a look at the KeyFrameDisplay class and followed the calculation there to get x,y,z.

NikolausDemmel, 2018-01-03:
Makes sense. Thanks.

mrwtakev, 2018-04-03:
@m2b821 Hello,I now need to get the depth map through the DSO, but I don't know how to change which parts of the code to implement，and I see that you have generated a depth map. Can you give me some advice? Thank you very much.

m2b821, 2018-04-03:
@mrwtakev You need to register an Output3DWrapper. An example is given in IOWrapper\OutputWrapper\SampleOutputWrapper. To just get the generated depth map: Use pushDepthImageFloat, which returns the inverse depth map in MinimalImageF, you can read it out using MinimalImageF->at(width, height). Also take a look at the SampleOutputWrapper. Remember to register any new output wrapper in the main: e.g. fullSystem->outputWrapper.push_back(new IOWrap::SampleOutputWrapper());
Hope this helps.

mrwtakev, 2018-04-03:
@m2b821 Thank you very much, I probably understand how to do it.But about'MinimalImageF->at(width, height)'I do not understand，especially at().Can you explain it?Thank you.