JakobEngel/dso #111: static_assert failed "Allocator::value_type must be same type as value_type"

mrwtakev, 2017-11-30:
When I build the DSO,there is an error occur, it is shown as following:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/map:820:5: error: 
      static_assert failed "Allocator::value_type must be same type as value_type"
    static_assert((is_same<typename allocator_type::value_type, value_type>::value),
    ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/charlie/dso/src/OptimizationBackend/EnergyFunctional.h:121:6: note: in instantiation of template
      class 'std::__1::map<unsigned long long, Eigen::Matrix<int, 2, 1, 0, 2, 1>, std::__1::less<unsigned
      long long>, Eigen::aligned_allocator<std::__1::pair<unsigned long long, Eigen::Matrix<int, 2, 1, 0,
      2, 1> > > >' requested here
          > connectivityMap;
            ^
6 warnings and 1 error generated.
make[2]: *** [CMakeFiles/dso.dir/src/FullSystem/FullSystem.cpp.o] Error 1
make[1]: *** [CMakeFiles/dso.dir/all] Error 2
make: *** [all] Error 2

NikolausDemmel, 2017-11-30:
Which compiler version is this?

Reminds me of https://github.com/JakobEngel/dso/pull/98#issuecomment-337342171, where downgrading to an earlier xcode version solved the issue. Would be nice to resolve it properly though.

In fact searching for this error reveals that maybe we are missing a const. Could you try changing the definition of `connectivityMap` in `EnergyFunctional.h` to the following (note the extra const):

```
	std::map<uint64_t,
	  Eigen::Vector2i,
	  std::less<uint64_t>,
	  Eigen::aligned_allocator<std::pair<const uint64_t, Eigen::Vector2i>>
	  > connectivityMap;
```

NikolausDemmel, 2017-12-01:
Can you test #113, please? With that, it should build with xcode 9 as well.

mrwtakev, 2017-12-02:
Thank you very much,it build with Xcode well now.

NikolausDemmel, 2017-12-02:
👍, thanks for testing.

PR #113 (2017-12-04): Fix declarations of aligned allocators.
Fixes #111. It seems like you need a `const` for the first argument of a
`std::pair` in `aligned_allocator` to make xcode 9 happy.

Fix the override of `PangolinDSOViewer::publishGraph`. Without this patch, the display of constraints does not work as it used to.

Added some `override` statements to Output3DWrapper's derived classes, to avoid such problems in the future.
closed 2017-12-04 by NikolausDemmel