JakobEngel/dso #114: Can the generated result file be in PCD format?

mrwtakev, 2017-12-04:
Hello，
I successfully run this program and found that the generated point cloud file is txt format,can the generated result file be in PCD format?

NikolausDemmel, 2017-12-04:
I'm not sure saving the pointcloud to a text file is implemented, or is it? And also not pcd. But PR's are welcome.

mrwtakev, 2017-12-04:
I see the result file is Result.txt, how can I convert its output format? For example, PR?

NikolausDemmel, 2017-12-04:
`result.txt` contains camera poses, as described in the readme. I suggest to also check the code to see whats implemented.

PR means pull request, i.e. you are welcome to implement this yourself and provide a pull request to merge it back for the benefit of others. It also means that no-one is currently planning to implement this.

closed 2017-12-07 by NikolausDemmel