JakobEngel/dso #115: Could it be possible to make a real-time-streaming DSO?

sun0shine, 2017-12-05:
Hello guys, I already run DSO with dso_dataset launchfile and provided datasets like sequence_01, sequence_02...
But the issue what i want is running dso in real-time-streaming.
Can i make it while i'm recording with my smart-phone? or Is there anybody going on like this methods?

NikolausDemmel, 2017-12-07:
DSO implements linearized, single-threaded processing, as well as multi-threaded "real time" processing. Check out https://github.com/JakobEngel/dso_ros for an example of how to use DSO with live camera input.

sun0shine, 2017-12-18:
@NikolausDemmel Thanks :)

closed 2017-12-31 by NikolausDemmel

lcqwe, 2018-04-29:
Is there any method for making a real-time-streaming DSO on windows with a mono camera. or says that How can I use no-ros wrapper for this thing?

NikolausDemmel, 2018-04-29:
@lcqwe, you will probably have to write your own wrapper that receives the images somehow (e.g. camera driver) and passes them on to DSO. You can take the ROS wrapper as an example of what to do.