JakobEngel/dso #118: Is it possible to focus on a selected area ?

LourensPool, 2017-12-22:
Is it possible to focus on just a selected area within the video ?

NikolausDemmel, 2018-01-03:
I'm not sure exactly what you mean, but I guess you want to crop the image before any further processing? I don't think cropping to an arbitrary region as preprocessing is supported, but you can specify arbitrary desired camera parameters for the rectification process. Otherwise, I guess the quickest solution might be to crop the images outside DSO as a preprocessing step.

s3r637, 2018-01-04:
Hi, how about to use a (handcrafted) vignette to mask out unwanted regions. Dirty hack! But to get a first impression...

Cheers

NikolausDemmel, 2018-01-04:
Sure, the is a simple one to try, however I suspect DSO might pick up the boundary as possible high-gradient feature points, which might corrupt the result.

If the ROI is not rectangular, then extending the code to load an additional mask and then never selecting points in that region is probably the way to go.

LourensPool, 2018-01-08:
The vignette is indeed what I wanted to try @s3r637 . And DSO indeed picks up the boundary points of the crop which corrupts the results. 

Never selecting points in that region is the way to go. Documentation of the code is sometimes tricky to read. 

Which class / function is best suited to add the " de selection" of the feature points ?

s3r637, 2018-01-08:
Hi @LourensPool, I had the same problem. We mounted a cam on a car roof and the whole engine cover was visible in the image. Black pixels -- means the camera is blind in this region -- over the whole engine cover in the vignette (not in the input images!) fixed the issue. Not one boundary point was selected.

cheers