JakobEngel/dso #119: Camera coordinate system from DSO

m2b821, 2018-01-16:
Hello,
I have a small question: Is the returned camera pose (R, t) for calculating the transformation into the camera coordinate system or inverse?
So is `x_ccs = R*x_gcs+t` or `x_ccs=(x_gcs-t)*inv(R)` 
with x_ccs: point in camera coordinate system and x_gcs: point in global coordinate system.
Thank you!

NikolausDemmel, 2018-01-18:
Depends on which variables you look at. Internally `worldToCam` is estimated, which transforms points from world coordinates into camera coordinates (i.e. `x_cam  = R_worldToCam * x_world + t_worldToCam`). But also variables that store `worldToCam` there. Also in the paper the notation `T_i` is used for the transformation `world_To_Cami`.

Note that in your example the inverse transformation is wrong. It should be `x_ccs=inv(R)*(x_gcs-t)`.

m2b821, 2018-01-23:
@NikolausDemmel:  Oh, I forget. Just to be clear: the returned data from publishDepthImage for example is in the camera coordinate system or already in the world coordinate system?

NikolausDemmel, 2018-01-23:
Did you have a look at the comments and the implementation? E.g. here:

https://github.com/JakobEngel/dso/blob/e59db70710ead30d6b47959f886665ddd9cb4f85/src/IOWrapper/Output3DWrapper.h#L178-L196

https://github.com/JakobEngel/dso/blob/e59db70710ead30d6b47959f886665ddd9cb4f85/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h#L125-L147

A depth map is naturally in image coordinates. I'm not sure what a depth map in world coordinate system would look like...

closed 2018-01-25 by m2b821