JakobEngel/dso #123: Assertion `std::isfinite(p->data->step)' failed   on my own dataset

dymymao, 2018-02-09:
when used on my own dataset , assert failure occurs frequently but not the tracking LOST
preset=2/3
-------------------------------------------------------------------------------------------------------------------------------
dso_dataset: /home/manco/workspace/slam/dso/src/OptimizationBackend/EnergyFunctional.cpp:317: void dso::EnergyFunctional::resubstituteFPt(const VecCf&, dso::Mat18f*, int, int, dso::Vec10*, int): Assertion `std::isfinite(p->data->step)' failed.
-----------------------------------------------------------------------------------------------------------------------------
so anyone can help me ?  Thanks a lot!

farhanmubasher, 2018-05-24:
Hi @dymymao ,Do you find any solution of it ? I'm continuously seeing this error.

ToniRV, 2019-06-04:
Hi, I see the same issue, for TUM mono datasets as well.

sanderisbestok, 2019-06-05:
Hi have the same issue for both TUM and other datasets. Not always sometimes it does succeed. @JakobEngel

Victorrr1, 2024-03-12:
I met the same issue and i fix it after rm build and recompile, no idea why it occur though.