JakobEngel/dso #124: recipe for target 'bin/dso_dataset' failed

Tacode, 2018-03-10:
/usr/bin/ld: CMakeFiles/dso_dataset.dir/src/main_dso_pangolin.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
CMakeFiles/dso_dataset.dir/build.make:139: recipe for target 'bin/dso_dataset' failed
make[2]: *** [bin/dso_dataset] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/dso_dataset.dir/all' failed
make[1]: *** [CMakeFiles/dso_dataset.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

NikolausDemmel, 2018-03-10:
Which version exactly ist this (what branch)? Which OS?

Have you tried the changes from #82 and/or #109?

Tacode, 2018-03-11:
Yes，I have tried the changes form #82 .
This problem has been solved.

closed 2018-03-11 by Tacode

nagu0011, 2018-12-01:
[ 57%] Built target CsvImport
[ 58%] Built target QueryRunner
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_new@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_get_verify_mode@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_set_verify@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `CRYPTO_num_locks@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_set_default_passwd_cb_userdata@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_get_wbio@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `BIO_ctrl@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_peek@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `CRYPTO_set_dynlock_destroy_callback@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_use_certificate_chain_file@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_get_shutdown@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_free@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_get_error@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `CRYPTO_set_locking_callback@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `TLSv1_method@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_new@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_load_error_strings@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ASN1_STRING_data@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_get_error@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_NAME_get_entry@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_write@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `CRYPTO_free@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `EVP_cleanup@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_verify_cert_error_string@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_free@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_load_verify_locations@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ASN1_STRING_length@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_use_PrivateKey_file@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_peek_error@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_free_strings@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `sk_num@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_free@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_accept@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_remove_state@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_ctrl@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ASN1_STRING_to_UTF8@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_clear_error@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `TLSv1_2_method@OPENSSL_1.0.1'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_get_rbio@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_set_fd@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `CRYPTO_cleanup_all_ex_data@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `BIO_s_mem@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_read@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_set_bio@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `GENERAL_NAME_free@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `ERR_reason_error_string@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_connect@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_NAME_get_index_by_NID@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_ctrl@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `RAND_poll@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_set_default_passwd_cb@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSLv23_method@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `BIO_new@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_library_init@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_get_subject_name@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `SSL_CTX_set_cipher_list@OPENSSL_1.0.0'
/usr/local/mapd-deps/lib/libthrift.so: undefined reference to `X509_get_ext_d2i@OPENSSL_1.0.0'
[... cut ...]
Import/CMakeFiles/KafkaImporter.dir/build.make:108: recipe for target 'bin/KafkaImporter' failed
make[2]: *** [bin/KafkaImporter] Error 1
CMakeFiles/Makefile2:1354: recipe for target 'Import/CMakeFiles/KafkaImporter.dir/all' failed
make[1]: *** [Import/CMakeFiles/KafkaImporter.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
-- Configuring done
-- Generating done
-- Build files have been written to: /home/thickstat/Downloads/mapd-core/build
[ 58%] Built target rerun_cmake
[ 84%] Built target QueryEngine
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2