JakobEngel/dso #126: Energy functional code structure?

mpizenberg, 2018-03-17:
Hi, I'm trying to understand the energy minimization code but having a hard time doing so. From the paper, Gauss-Newton is used for the optimization problem. While reading the code, it seems that it is used at multiple places, epipolar search of immature points, estimation of rigid body motion and brightness transfer parameters in coarse tracking of new frame, and in joint optimization of active window. I'm not well versed in optimization so I'm having a lot of trouble understanding this last part. What are the roles of `AccumulatedSCHessian`, `AccumulatedTopHessian` etc. ?

Is there anyone who could provide a high level (beginner friendly) explanation of how the energy functional code works?

trulyspinach, 2019-12-05:
I am finding it hard to understand as well. Have you incase realized what are the roles of them?

DengueTim, 2025-09-06:
It's hard code to understand, especially because in the `Accumlated*` classes the code is optimised for speed not intelligibility.  I found [Tony Ling's DSO blog posts](https://tongling916.github.io/tags/#DSO) helpful as it breaks down the equations and how they are implemented.