JakobEngel/dso #128: About depth map

mrwtakev, 2018-04-03:
Hello,I now need to get the depth map through the DSO, but I don't know how to change which parts of the code to implement. Can you give me some advice? Thank you very much.

s3r637, 2018-04-26:
Hi @mrwtakev, 

write your own [Output3DWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Output3DWrapper.h), see also [SampleOutputWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h).

cheers

Sho412, 2019-06-22:
Hi,
I'd like to use depth info and keyframe info.
pushDepthImage() can get depth image. However I can't find out how to attach depth information to keyframe information. ex.  Relation between the point on the depth map and the keyframe id which has the point.

So, as the first step, I try to get point information from publishKeyFrames() and store in the PangolinDSOViewer.h. Then put the depth info on the raw image in pushLiveFrame().
It similar to the output of pushDepthImage() on the Pangolin. However slightly different.
Is this implementation shortcut toward my goal?