JakobEngel/dso #132: run error when using my own dataset

phenix1983, 2018-04-13:
Hi,
Thank you for your sharing, I want to run with my own dataset.
The input images are named like
"00000.jpg
00001.png
...
00202.png"
camera.txt constant is like bellow:
0.25114453125 0.44571944444444 0.4993335940000 0.501525000000000 0.000000000000000
1280 720
none
1280 720

input command is :bin/dso_dataset files=APA_video/1/images.zip gamma=APA_video/1/pcalib.txt preset=0 mode=2

question 1: whether I do have to input pcalib.txt, when mode =2? I don't know whether it is related to different cameras.
question 2: the error is in the 
[image]
 picture.

Thanks in advance.

haakonelf, 2019-04-09:
Hey! I am in the same situation, did you ever find out about it?

s3r637, 2019-04-15:
Hi, try to use none-normalized intrinsic values (use the entries of `K` directly).

Cheers