JakobEngel/dso #134: Pixel position of every possible candidate

ImmerHenry, 2018-04-17:
I ran the DSO with my own Dataset and it's quite promising. I want to try to combine some things with my own openMVG framework and see how this performs.  

What I need is the pixel position of every possible candidate in every image (like the green dots in the image).
[image]

The other information which I am curious is: Which candidate in image j is the same candidate in image j+1, j+2, j+x.
Do you think this information could be written into a txt-file?

Thank you in advance!

s3r637, 2018-04-26:
Hi @ImmerHenry,

sure! Write your own Output3DWrapper, see also SampleOutputWrapper.

Since you have the poses and calibration you just need to reproject the hosted keypoints of one frame to other frames. The candidates are more tricky, they don't have their final idepth yet, just make an average of `min_idepth` and `max_idepth` if you really need the candidates.

(However, to store this whole information into a file seems superfluous to me.)

cheers

ImmerHenry, 2018-05-02:
@s3r637 
Thx a lot! 
However, when I use  

> outfile<<fh->shell->trackingRef->id<<"  "; 

to export the KeyFrameID and use 

            for(ImmaturePoint* p : fh->immaturePoints)
            {
                pc[numSparsePoints].u = p->u;
                pc[numSparsePoints].v = p->v;

            outfile<<" the "<<numSparsePoints<<" points   ";
            outfile<<pc[numSparsePoints].u<<"  ";
            outfile<<pc[numSparsePoints].v <<"  ";

                numSparsePoints++;

            }
to export immaturepoints at the same time. It always appears segmentation fault( core dumped). Could you tell me what is probably wrong with that?

s3r637, 2018-05-03:
Hi @ImmerHenry, I believe the `trackingRef` means the predecessor frame (I'm unsure, I would have to look in the code..) and the first frame has none.
However, use `fh->shell->incoming_id` to get the KeyFrameID. It looks like you inject your code somewhere in the Pangolin wrapper!? Your should better write you own Output3DWrapper, like:

```
void publishKeyframes(std::vector<dso::FrameHessian*> &frames, bool final, dso::CalibHessian* calib) override
{
   if (final) {
      for (dso::FrameHessian* keyframe : frames) {
         if (keyframe->shell->poseValid) {
            doStuff(keyframe, calib); // your business logic function
         }
      }
   }
}
```

hope it helps

ImmerHenry, 2018-05-03:
@s3r637 Thx! 

It works for the keyframes. What's more, do you know are there any ways to export all the points from not only keyframes but also Nonkeyframes? E.g. in the traceNewCoarse function, or any other functions?

Thank you in advance.

s3r637, 2018-05-07:
Hi @ImmerHenry,

you really should take a look at  [Output3DWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Output3DWrapper.h) and [SampleOutputWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h)...

```
void publishCamPose(dso::FrameShell* frame, dso::CalibHessian* calib) override
{
   if (frame->poseValid) {
      doStuff(frame, calib); // your business logic function
   }
}
```

Be aware, `publishCamPose` will also called for keyframes. (See also [Issue#136](https://github.com/JakobEngel/dso/issues/136).)

cheers