JakobEngel/dso #135: How to measure distance using poses (3x4 matrix returned by camToWorld)

farhanmubasher, 2018-04-23:
Hi all, 
I'm new to dso, I have got poses of camera (in 3x4 matrix) using camToWorld. But i don't know how to use these poses for further process. How can i use these poses to measure distance in meters covered by my camera?  This is the sample output of poses, how can i use to measure distance?
[0.986279  -0.0681795   -0.150348   0.0170084
   0.069339     0.99759  0.00247696 -0.00431342
   0.149817  -0.0128679     0.98863    0.406184 ]
Thanks.

farhanmubasher, 2018-04-26:
Thank you very much @s3r637  . 
I appreciate your time :)

closed 2018-04-26 by farhanmubasher

brandnew2017, 2019-05-28:
@farhanmubasher Hi, I am also new to dso, and I want to know how you get poses of camera?
Thank you!

DRAhmadFaraz, 2019-10-17:
@brandnew2017 Have you able to get the poses of every frame.?

NikolausDemmel, 2019-10-17:
Please stop spamming people.