JakobEngel/dso #136: How can I save trajectory of ALL frames

JeffreyYafeiHu, 2018-04-25:
Hi JaKob, 

I would like to ask how can I save trajectory of all frames instead of just keyframes?

Thanks you!

070411209, 2018-04-25:
Hi，Jeffrey,

I am writing a little of blog about slam, and encounter some problems.

No matter keyframes or trajectory data, all for Pose optimization or where are camera.

if you want save trajectory data of all frames, first, you should konw what to save.

second,  you should know What data structure is used.

last, answer the above two question.

thanks you.

***** 王包东 *****
**  15507553214 **
*nmgwbd@163.com*
__________________________
 
From: Jeffrey Hu
Date: 2018-04-25 15:25
To: JakobEngel/dso
CC: Subscribed
Subject: [JakobEngel/dso] How can I save trajectory of ALL frames (#136)
Hi JaKob,
I would like to ask how can I save trajectory of all frames instead of just keyframes?
Thanks you!

s3r637, 2018-04-26:
Hi @JeffreyYafeiHu,

write your own [Output3DWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Output3DWrapper.h), see also [SampleOutputWrapper](https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h).

Cache all poses of all frames in, e.g., a `std::map<Frame ID, Frame Pose>`. Keep your Frame Poses up to date (on function `publishKeyframes` and `publishCamPose`) and dump the `std::map` to a file at the end (use the `join` and/or the `reset` function).

cheers

madmage, 2018-08-07:
Dear @s3r637, thanks for the response to @JeffreyYafeiHu, I tried to implement your advice, but I have one question: once a keyframe is updated (i.e., publishKeyframes is called), should I also update all the poses related to this Keyframe?

s3r637, 2018-09-26:
Hi @madmage,

what do you mean by "all the poses related to this Keyframe"? A Keyframe has only one (absolute) pose (related to the first Keyframe).

cheers

DRAhmadFaraz, 2019-10-16:
@JeffreyYH @070411209 @s3r637 @madmage 

I want the camera poses of every frame in results.txt so can you please provide me the changes in code.?

I will be thankful to you.

s3r637, 2019-10-17:
Hi @DRAhmadFaraz,

please take a look into the provided [README ](https://github.com/JakobEngel/dso/blob/master/README.md#34-accessing-data)file and going through the existing issues should also very useful to reach your goal faster.

cheers