JakobEngel/dso #137: bug in CoarseInitializer::doStep

kokerf, 2018-04-27:
Hi JaKob,
    I wonder whether there is a bug in [CoarseInitializer::doStep](https://github.com/JakobEngel/dso/blob/master/src/FullSystem/CoarseInitializer.cpp#L895).

the code is:
```c++
		float b = JbBuffer[i][8] + JbBuffer[i].head<8>().dot(inc);
		float step = - b * JbBuffer[i][9] / (1+lambda);
```
I think it should be '/' rather than '*' in the second line:
```c++
                float step = - b / JbBuffer[i][9] / (1+lambda);
```
we can state the system as:

[image]

the $\delta \xi$  contain the pose and affine parameters which is sloved(`inc`) by Schur complement ahead, and we want to calculate the inverse depth $\delta d$. I slove the answer as:

[image]

in [here](https://github.com/JakobEngel/dso/blob/master/src/FullSystem/CoarseInitializer.cpp#L445) and [here](https://github.com/JakobEngel/dso/blob/master/src/FullSystem/CoarseInitializer.cpp#L927) we get correspondences:

[image]

So there should be the inverse of `JbBuffer[i][9]`. Look forward to your answer.

blindnote, 2018-04-30:
It's inversed [here](https://github.com/JakobEngel/dso/blob/e59db70710ead30d6b47959f886665ddd9cb4f85/src/FullSystem/CoarseInitializer.cpp#L555)

kokerf, 2018-05-01:
@blindnote thank you! I found it!

closed 2018-05-01 by NikolausDemmel