JakobEngel/dso #142: How to build map and reuse it using DSO

farhanmubasher, 2018-05-09:
Hi all..
I'm using dso_ros for real time tracking and i'm getting poses from  camToWorld.data() ( e.g x,y,z,  qx, qy, qz, qw). But how can i build map using these poses and reuse that map.
For exmaple, I want to move a robot in a square. So firstly i will move it manually in a square so that it can build a map and after that it should move in that map autonomously .
* is there any other way to do it ? (Build map from unknown environment using dso and then reuse it so robot can move autonomously in that involvement).

Thanks in advance !!

NikolausDemmel, 2018-08-15:
duplicate with https://github.com/JakobEngel/dso_ros

anika001, 2020-01-23:
Hi all, 

@farhanmubasher  I am trying to do the same thing/ saving the map and then launching DSO on localization mode so that the robot/vehicle can move in the map . 
Did you manged to do it ? could you give me any hints if possible? 

@NikolausDemmel  could you please give me the issue which deals with this in dso github, I did not find anything related to this 

thank you all in advance