JakobEngel/dso #146: Include <ros/ros.h>

bjbraun, 2018-05-24:
Hi,

how is it possible to include ros? I have tried changing the cmake file and package file by including the following changes (I have used the cmake branch of Nikolaus Demmels Version of the DSO https://github.com/NikolausDemmel/dso/tree/cmake):

Cmake-File:

find_package(catkin REQUIRED COMPONENTS)
find_package(roscpp)
catkin_package(CATKIN_DEPENDS roscpp)
include_directories(
  ${catkin_INCLUDE_DIRS}
)
target_link_libraries(dso_dataset
   ${catkin_LIBRARIES}
 )

Package:
<buildtool_depend>catkin</buildtool_depend>
<depend>roscpp</depend>

It is building including these changes (using catkin build).
However, if i then try to add "#include <ros/ros.h>" to any file, it says that it can't finde the file.
My goal is it, to publish some values in the SampleOutputWrapper.

Everything else is working perfectly fine.

Thank you really much in advance for your help!

NikolausDemmel, 2018-05-25:
You would like to use ros directly in the DSO package and not dso_ros?

You need to adapt the package.xml and CMakeLists.txt.

Best remove these lines:
https://github.com/NikolausDemmel/dso/blob/5c77bcca3ca6b81abd6737f967f28442ec735317/package.xml#L18-L20

Also, in cmake lists you should do:

```
find_package(catkin REQUIRED COMPONENTS roscpp)
```

bjbraun, 2018-05-25:
Hi Nikolaus,

it is working this way, thank you really much!

Is there a better way that I can access the data in ros, for example from the function publishKeyframes?
I tried to use dso_ros, which is basically working. However, I was not able to get the data from for example publishKeyframes to the main.cpp of dso_ros so that I can publish them there.

NikolausDemmel, 2018-05-26:
> Is there a better way that I can access the data in ros?
I tried to use dso_ros, which is basically working. However, I was not able to get the data from for example publishKeyframes to the main.cpp of dso_ros so that I can publish them there.

You are free to extend the output wrapper interface if it doesn't give you all the info you need. But I guess it is also fine to add ROS publishers directly to DSO if that makes your life easier, but that is less modular in the long run.

amjack0, 2020-07-13:
dfdf

zeeshanhabibvirani, 2020-08-14:
hey @bjbraun , Can you share your SampleOutputWrapper file. I also want to publish data from there but I cant write publisher in it.

chapa17, 2021-11-01:
Hello @bjbraun 

I am getting the same error of ros.h not found. I already cross checked CMakeLists.txt and package.xml but they seem fine. Is there any additional step you modified to make it work except the suggestion provided by @NikolausDemmel?

Thank you!

chapa17, 2021-11-02:
Hello @bjbraun 

I am also trying to run dso directly with ros. I am able to compile my OutputWrapper although when I try to run the code I get an error that `Couldn't find executable named dso_dataset below /home/xxx/catkin_ws/src/dso.`

Can you please let me know the directory path from where  you executed the `rosrun` command?

Thank you!