JakobEngel/dso #148: Is there an available code for VI-DSO? (DSO combined with IMU)

ronir121, 2018-05-31:
The paper for visual-inertial odometry looks very interesting, is there an available source code for it?
( https://vision.in.tum.de/research/vslam/vi-dso )

NikolausDemmel, 2018-05-31:
Not for the moment unfortunately. It is unclear if and when it might be published.

garethblock, 2019-01-22:
Hi @NikolausDemmel, agreed with ronir121 -- the VI-DSO looks like a great addition.  I would really like to run the source code on our datasets... is it possible to provide access?  Thanks!

NikolausDemmel, 2019-01-23:
It's not in my hands and as far as I know it is not possible. Sorry.

I'll close the issue since I guess it has been answered.

closed 2019-01-23 by NikolausDemmel

lukasvst, 2022-05-11:
A (late) update on this: An official implementation for the successor of VI-DSO is now available at: https://github.com/lukasvst/dm-vio