JakobEngel/dso #153: FPS evolution with higher frames

caap95, 2018-07-12:
Hi,

I'm a finalist student and for my thesis I'm working on various SLAM algorithms and studying their performance. One test that I made was to check if the FPS would vary for datasets with different number of frames.

For DSO I got the following result:

[image]

I thought that the explanation would be because before it stabilizes it didn't yet created the 7 keyframes that the algorithm uses to compare the new points and decide the new keyframes. So, because in the beginning it makes less comparisons is faster. Although, with further observations i saw that, for the dataset that i tested, it takes only around 200 frames to create the seven keyframes. So my interpretation is not correct.

Can you help me understanding this result? 

Thanks.

NikolausDemmel, 2018-07-29:
One thought: The number of keyframes / second is strongly correlated with FPS, since every keyframe causes the backend solver to run. The number of keyframes per second depends strongly on the motion the camera makes. If you move fast and to new places, many keyframes are created. If you move slow or keep the camera focused on the same thing, fewer keyframes are created. You should maybe also plot "keyframes per second" vs "no of frames".

NikolausDemmel, 2018-08-18:
Thanks for the followup and plots.

closed 2018-08-18 by NikolausDemmel

caap95, 2018-08-18:
@NikolausDemmel, sorry to insist. But not that I analyze more carefully this data there's one thing I can't understand. This graphics are great and explain the majority of the behavior, but how can we explain that the final FPS (after stabelization) is bigger than the initial FPS, taking into account that, from the plot to the right, we have more keyframes being creating per second?

caap95, 2018-08-18:
Hi @NikolausDemmel, thanks again for your quick answers. I did as you suggested, but I think the results are a bit contraditory. Here are the graphs:

[image]
[image]

I got the following data for each keyframe: Number of iterations on the optimization, number of points to optimize and total optimization time. The values on the graphs are a mean of values, I know that for dataset with 100 frames I got 4 keyframes, for 200 frames 6 keyframes, for 400 30 keyframes, etc etc.. I did the mean of the respective keyframes for each dataset. 

The first graph, number of iterations, makes sense to explain our previous results. Although the others don't. 

What do you think of this? 

Thanks