JakobEngel/dso #155: Pangolin shows only first few frames and then stops

Nihaar1996, 2018-07-17:
[image]

Pangolin initialises and displays few frames (say about 10) and then just freezes, as if it is waiting for something to happen. There is no error thrown in the terminal output just that it stops at the place shown in the attahced figure.

This is for a mac. Any help is greatly appreciated!

NikolausDemmel, 2018-07-18:
Hmmm, not sure. I think I might have seen that before.

Some thoughts:

 - maybe try the `nomt=1` and `speed=0` options.
 - try to debug / add some output to check where exactly it is blocking.
 - try on Ubuntu, if possible

Nihaar1996, 2018-07-18:
Oh that's super advice, @NikolausDemmel that worked! Many, many thanks!

closed 2018-07-18 by Nihaar1996

whubaichuan, 2020-09-23:
@nihaarshah Please use Linux. Many problems in mac will not exist in Linux.