JakobEngel/dso #157: Output image sizes

ImmerHenry, 2018-07-23:
I ran the DSO with my own Dataset and it's quite promising. 

During several changes of every parameters, I found DSO performs best and the pointclouds are the most accurate if my images output size are the multiple integer of 16. Like 640*480 or 1120* 800.

Could anyone please help me explain that? I am really curious about that.

NikolausDemmel, 2018-07-23:
Interesting. I just have some vague guesses, which would have to be investigated:

 - Some performance optimizations might be tuned for multiples of 16. This would make a difference if you run in real-time (not linearized) mode.
 - Bucketing in point selection might be tuned for multiples of 16, although I would suspect this to make not too much a difference tbh.

AlexandrGraschenkov, 2018-07-26:
In DSO used pyramid of images and tracking working over all images in pyramids. So when you use size power of 16, it better work with border pixels. Otherwise it select pixels near border in outside the memory of image and try track it.

NikolausDemmel, 2018-07-29:
Oh yes, I remember there were some issue with pixel selection and access outside the image bounds. It would be great to fix those bugs...

AlexandrGraschenkov, 2018-07-30:
It's not hard to fix it
#158

NikolausDemmel, 2018-07-30:
Thanks @AlexandrGraschenkov. 

@ImmerHenry, could you try that PR and check whether that fixes the performance discrepancies between different resolutions that you observed?

ImmerHenry, 2018-07-30:
@NikolausDemmel , thanks a lot for your help. I will have a try on it.

ImmerHenry, 2018-07-30:
@NikolausDemmel @AlexandrGraschenkov , I tried with that PR, but the performance discrepancies between different resolutions still exists. Moreover, I changed the images output resolution, not the input resolution, so is it still related to window of 32x32 pixels?

AlexandrGraschenkov, 2018-07-30:
@ImmerHenry maybe you lost some info about the image during resize? If you have image 960x720 and resize to 640x480, then 3 pixels will map to 2 pixels. During this mapping you interpolate color of pixels and lose info. With algorithm like DSO we loose a lot of information.

What about PR: you can look on debug Pangolin output and check selecting pixels near the border. Now near bottom border noise not select by algorithm. I'm not sure about `width` not dividable by 32. Not tested yet.