JakobEngel/dso #162: How to publish pointclouds in ros?

sammyjan, 2018-09-26:
I want to publish pointcloud,camera pose and some other data in ROS, to make them can be visualised in the Rviz. I have read the codes for a few days, but I still don't have any ideas.

s3r637, 2018-09-26:
Hi @sammyjan,

I assume that this is not your first ROS assignment...
Therefore, the following link should be a good starting point for you to master the task without any hiccups.

https://github.com/JakobEngel/dso/blob/master/README.md#34-accessing-data

Additional links help you reach the goal faster.

https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Output3DWrapper.h
https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/OutputWrapper/SampleOutputWrapper.h
https://github.com/JakobEngel/dso/blob/master/src/IOWrapper/Pangolin/PangolinDSOViewer.h
https://github.com/JakobEngel/dso/blob/master/src/main_dso_pangolin.cpp#L407

Furthermore, you are not alone ... look at the past issues.

https://github.com/JakobEngel/dso/issues?utf8=%E2%9C%93&q=ROS
https://github.com/JakobEngel/dso/issues?utf8=%E2%9C%93&q=point+cloud+
https://github.com/JakobEngel/dso/issues?utf8=%E2%9C%93&q=pose

Cheers & happy coding!

sammyjan, 2018-09-27:
@s3r637 Thank you for your detailed advice, I'm sure it'll help me a lot. Thanks again!

sammyjan, 2018-10-18:
@s3r637 Hello. Thanks for your help, I can publish pointcloud and pose in the Rviz now.But the pointcloud map is skew in the Rviz. How can I solve it?

Hydrorion, 2018-10-25:
Hello,
it's my first project on ROS (maybe it's not the best way to start) and i want to implement DSO in my project.
@sammyjan how did you manage to get those pointcloud ?  The file Output3DWrapper.h doesn't explain what those functions does, only when their are used so i'm still confuse. Whitch object contain the hessian point vector ?

sammyjan, 2018-10-26:
Hello @Hydrorion , I modify the code on the basis of the package from [https://github.com/Neoplanetz/dso_with_saving_pcl](url). What's more,This issue [https://github.com/JakobEngel/dso/issues/55](url) may help you a lot.To be honest , I am a beginner, so I am not clear about some details . I modify the SampleOutputWrapper.h , I show a part of the code as follows, you can take a reference. Hope you can reach your goal successfully. Cheers!
`

namespace dso
{

class FrameHessian;
class CalibHessian;
class FrameShell;

namespace IOWrap
{

class SampleOutputWrapper : public Output3DWrapper
{
public:
        inline SampleOutputWrapper()
        {
            numPCL = 0;
            isSavePCL = true;
            isPCLfileClose = false;            

            pclFile.open(strTmpFileName);

            printf("OUT: Created SampleOutputWrapper\n");
        }

        virtual ~SampleOutputWrapper()
        {
            if (pclFile.is_open())
            {
                pclFile.close();
            }

            printf("OUT: Destroyed SampleOutputWrapper\n");
        }

        virtual void publishKeyframes( std::vector<FrameHessian*> &frames, bool final, CalibHessian* HCalib) override
        {
            float fx, fy, cx, cy;
            float fxi, fyi, cxi, cyi;
            //float colorIntensity = 1.0f;
            fx = HCalib->fxl();
            fy = HCalib->fyl();
            cx = HCalib->cxl();
            cy = HCalib->cyl();
            fxi = 1 / fx;
            fyi = 1 / fy;
            cxi = -cx / fx;
            cyi = -cy / fy;
             
            if (final)
            { sensor_msgs::PointCloud cloud;
              cloud.header.frame_id = "world";
              
                for (FrameHessian* f : frames)
                {
                    if (f->shell->poseValid)
                    {
                        auto const& m = f->shell->camToWorld.matrix3x4();
      
                        // use only marginalized points.
                        auto const& points = f->pointHessiansMarginalized;

                        for (auto const* p : points)
                        {
                            float depth = 1.0f / p->idepth;
                            auto const x = (p->u * fxi + cxi) * depth;
                            auto const y = (p->v * fyi + cyi) * depth;
                            auto const z = depth * (1 + 2 * fxi);
                       
                            Eigen::Vector4d camPoint(x, y, z, 1.f);
                            Eigen::Vector3d worldPoint = m * camPoint;

                            if (isSavePCL && pclFile.is_open())
                            {
                                isWritePCL = true;

                                pclFile << worldPoint[0] << " " << worldPoint[1] << " " << worldPoint[2] << "\n";

                                printf("[%d] Point Cloud Coordinate> X: %.2f, Y: %.2f, Z: %.2f\n",
                                         numPCL,
                                         worldPoint[0],
                                         worldPoint[1],
                                         worldPoint[2]);
                        geometry_msgs::Point32 h;
                        h.x = worldPoint[0];
                        h.y = worldPoint[1];
                        h.z = worldPoint[2];
                        cloud.points.push_back(h);                                               
                                numPCL++;
                             isWritePCL = false;                  
                            }
                             
                            else
                            {
                                if (!isPCLfileClose)
                                {
                                    if (pclFile.is_open())
                                    {
                                        pclFile.flush();
                                        pclFile.close();
                                        isPCLfileClose = true;
                                    }
                                }
                            }
                 cloud_pub.publish(cloud);
                         }
                
                    }
              
                }
            }
`

EdwinvanEmmerik, 2021-03-19:
@sammyjan Would it be possible to share your outputWrapper for ROS-rviz implementation?