JakobEngel/dso #163: How to get the direction vector for each frame?

Ascrin96, 2018-09-27:
Hello, dso is the very cool library, but  I want to position the camera on a pre-recorded cloud of points and for this I think if I get the direction vector for each frame and I know  descriptors of feature points. Maybe I'd be able to get position on the pre-recorded cloud. 
I wand to find the direction vector and descriptors of feature points and where they are situated in src of dso? 
Thank you for your suggestion.

s3r637, 2018-09-27:
Hi @Ascrin96,

what do you actually recorded? An image withe corresponding depth per pixel (e.g., with a stereo camera or kinect or similar)?

cheers

Ascrin96, 2018-09-27:
Hi @s3r637 , thank you for responding. 
No, I use ordinary camera. I saw https://github.com/Neoplanetz/dso_with_saving_pcl and added recorder trajectory on recorded trajectory on points: 
           f->shell->camToWorld.translation()[0];
	   f->shell->camToWorld.translation()[1];
	   f->shell->camToWorld.translation()[2];
And I have 2 pcl files. First is cloud of points and second points trajectory.

s3r637, 2018-09-27:
What is your target?
What do you mean by "I want to position the camera on a pre-recorded cloud of points"? Do you wanna run DSO a second time and use somehow the pre-recorded point cloud from the first run?

However, to get the direction vector for each frame you could use the camera poses. I think you need not only the translation part but also the orientation part to know how the camera is orientated in the scene.
[Issue#135](https://github.com/JakobEngel/dso/issues/135) should help you.