JakobEngel/dso #165: Is there a way to inject groundtruth into DSO?

MoBaT, 2018-10-15:
I want to do something similar to what this guy was trying to do:

https://robotics.stackexchange.com/questions/15456/how-to-properly-initialize-every-new-pose-in-a-visual-slam-algorithm-namely-dso

Is it possible? I've been attempting to modify the FullSystem to get it to inject my own poses but I could not do it. The only thing I was able to do was inject my pose when the system resets.

s3r637, 2018-10-21:
Hi @MoBaT,

inject your GT in the [lastF_2_fh_tries](https://github.com/JakobEngel/dso/blob/0ab8018028eff272591861921bef455d28435db7/src/FullSystem/FullSystem.cpp#L286) object. Be aware it requires an inverse (delta) motion (changes).

cheers

M1234Thomas, 2019-06-13:
hey @MoBaT ,

I am also facing the same problem. 

Did you manage to fix this? Could you give some directions in that case. 

Thanks in advance

M1234Thomas, 2019-06-13:
hey @MoBaT 

I am also facing the same problem.

Did you manage to fix this? Could you give some directions in that case.

Thanks in advance