JakobEngel/dso #167: How is the initialization setting the initial position

Metalzero2, 2018-10-30:
Hello,

I got DSO up an running with my own camera. What I do not understand is how the initial position is selected. Usually, when applying SLAM, the camera frame start from translation = (0,0,0) and rotation = (0,0,0). In my case, both these start with different values every times. So far it seems just random to me.

How is this initial position selected?

studennis911988, 2019-05-21:
The initial position of visual odometry depends on the time it finish initialization, so if you need global position information, VO is not enough for your need.

Metalzero2, 2019-05-22:
If I understand correctly, the time the initialization phase takes, will determine what it sets as its "home" position?

studennis911988, 2019-05-22:
@Metalzero2 Yes, you are right, so basically we can't determine the home position (0,0,0) in visual odometry.  By the way, are you still working on DSO? What project are you working on?

Metalzero2, 2019-05-28:
@studennis911988 

OK I see, that makes sense. 

I am not currently working on the project that I used it in. I was using the RealSense camera to perform handheld SLAM.