JakobEngel/dso #169: How to avoid parameter drift during state estimation?

vonzy, 2018-11-28:
Hello every one, recently I try to rewrite DSO, and I almost finish the nonlinear optimization part.  I found  the state estimates (especially affine brightness parameters) are easily to drift in my implementation while they are much more stable in original DSO. As a result, the tracking part is easily to fail because the affine brightness parameters are easily to exceed the threshold. I wonder how original DSO can have much more stable performance, can anyone help me out? Thank you very much!  
[image]

vonzy, 2018-11-30:
I have found some tricks in DSO that keeps system consistency: 
1. use FEJ ( the jacobian of intensity does't need to be fixed)
2. set huge prior for the states of the first frame.
3. set lower factor for marginalization to avoid inaccurate prior
4. orthogonalize the system equation
But I am not sure I am right, or there may be other methods I haven't found. Is anyone working on this too, maybe we can share some ideas?

DengueTim, 2025-09-06:
I'm surprised you have managed to rewrite the optimizer without understanding how the orthogonalizing code works! AFAIK, it adds perturbations in the unobservable parameter space(absolute rotation, scale etc..). This is then used to remove any drift in those directions from the update steps `EnergyFunctional::orthogonalize(VecX* b, MatXX* H)`.  Although I'm not an expert on linear algebra.