JakobEngel/dso #170: build

Myasnikova, 2018-11-29:
Hello! I have some problem with project building. I tried to build it on virtual machine ubuntu 18.04  following readme and after 95% of loading computer froze without any errors. I tried to do this three times and in every my attempt it froze. May be somebody had the same problem and will be able to help me, please.

NikolausDemmel, 2018-11-29:
Out of memory?

Myasnikova, 2018-11-29:
After stopping of build process I have 17Gb of free memory on my virtual machine and I didn't have any errors about lack of memory

NikolausDemmel, 2018-11-29:
Other than out of RAM I don't know of a reason why compiling should kill your VM, sorry...

Myasnikova, 2018-11-29:
RAM - 4 Gb, this is not enough?

Myasnikova, 2018-11-29:
What are the system requirements for this project?

NikolausDemmel, 2018-11-29:
4GB might not be enough. Watch your system monitor to see if OOM is the issue.

Things you can try:
 - more RAM for the VM
 - compile with `-j1`
 - try clang instead of gcc

kckwan, 2018-12-05:
I had same problem before. I built the DSO code with VM ubuntu in VirtualBox, and the `make -j` process freeze without any errors after 80% to 90% progress.
Due to the `-j` command (in the installation guideline), `make` run all the jobs (i.e., compiling .cpp files) simultaneously.
Each compiling job requires around 20 to 60 second, and it is too heavy for my VM.
I tried `make` instead of `make -j`, and the problem solved.