JakobEngel/dso #171: why are affine light parameters so easily to drift?

vonzy, 2018-12-12:
I notice that if I remove the constrain for these 2 parameters ( set setting_affineOptModeA & B = 0) in DSO，then these 2 parameters drift severely. And if I use this constrain in my implementation, the solver can't converge to a G-N solution. I wonder why these two parameters are so difficult to fix?

vonzy, 2018-12-12:
@NikolausDemmel @JakobEngel

ZhouTangtang, 2018-12-27:
I have the same problem.  I try to use the photometric error model applied in DSO, but I find my implementation can't converge at all(through G-N method). While if I just abandon the (a, b) parameters, my result gets much better. 
I thought I need to set a priority to (a, b), and I add (a=0, b=0) as part of the residuals, giving huge weighting. But it doesn't work as I thought. 
Maybe I need to learn more about DSO and optimization to handle these two parameters. It's not so easy as I thought. 😢

vonzy, 2018-12-27:
I have solved this problem😅

NikolausDemmel, 2018-12-27:
> I have solved this problem😅

What made it work for you?

vonzy, 2018-12-28:
I set huge prior as DSO......