JakobEngel/dso #174: Unsolved external symbol with functions that has FrameHessian as parameter

practisebody, 2019-01-19:
I'm using Visual Studio 2017 to build. I got an error complaining missing this function "int __thiscall dso::PixelSelector::makeMaps(struct dso::FrameHessian const * const,float *,float,int,bool,float)". This is because FrameHessian was defined to be a struct, but in a few files, it was declared as "class FrameHessian" (e.g. in EnergyFunctionalStructs.h, PixelSelector2.h ...). Thus, the exports function signature is like this: "class dso::FrameHessian const * const", so could not be found by the compiler.

By changing all the declarations of FrameHessian to struct the problem is solved.

NikolausDemmel, 2019-01-19:
Thanks for reporting. You are right, it should be the same everywhere...

practisebody, 2019-01-25:
Follow up: The same thing happens with CalibHessian, sometimes it's declared to be class and sometimes struct, causing unresolved external symbol problem in Visual Studio.