JakobEngel/dso #175: divide by zero - impacts timing info only

earwole1, 2019-01-22:
While trying to assess multi-thread performance within my VM (Ubuntu + Ryzen 5), I noticed that the variable ``MilliSecondsTakenMT'' was defined as "inf". This was set by sInitializerOffset and before that, timesToPlayAt.

Ultimately, around line 433 in commit ae1d0b3d972367b0a384e587852bedcf5f414e69:

` timesToPlayAt.push_back(timesToPlayAt.back() +  fabs(tsThis-tsPrev)/ playbackSpeed;
`

When playbackSpeed is zero (observed with mode=0, preset=2), the estimate for MT will be "inf".  The simple change is:

` timesToPlayAt.push_back(timesToPlayAt.back() +  fabs(tsThis-tsPrev)/ (0 == playbackSpeed ? 1 : playbackSpeed));
`

NikolausDemmel, 2019-01-22:
Makes sense, but I guess this only affects output (and `preset 3` is non-multithreaded anyway, right?).

Not sure if

```cpp
timesToPlayAt.push_back(timesToPlayAt.back() + fabs(tsThis-tsPrev)/ (0 == playbackSpeed ? 1 : playbackSpeed));
```

is what we want, maybe rather, 

```cpp
timesToPlayAt.push_back(timesToPlayAt.back() + (0 == playbackSpeed ? 0 : fabs(tsThis-tsPrev)/playbackSpeed));
```

but I suppose it doesn't matter, since with `0 == playbackSpeed` we only use it to set `sInitializerOffset` anyway. However, you need to check what happens with that.

Can you send a PR? 

Maybe double check that the output still makes sense for `0 == playbackSpeed`, since in 

https://github.com/JakobEngel/dso/blob/ae1d0b3d972367b0a384e587852bedcf5f414e69/src/main_dso_pangolin.cpp#L540

we use it as a starting point through `sInitializerOffset`, set in 

https://github.com/JakobEngel/dso/blob/ae1d0b3d972367b0a384e587852bedcf5f414e69/src/main_dso_pangolin.cpp#L461

however the way it is used, it seems to me that for `playbackSpeed = 0` we'd want to set it to the actual image read time for the first image after the initializer was successful?

earwole1, 2019-01-22:
I'll submit a PR this weekend