JakobEngel/dso #179: formulars in immaturePoint traceOn()

tanjunyao7, 2019-02-10:
Hi there,
I just can"t figure out how are some variables in the member function traceOn() of class immaturePoint calculated? For example errorInPixel, idepth_min and idepth_max. Can anybody give me some instructions or reference paper? thanks.
best

NikolausDemmel, 2019-02-16:
I'm not sure about the code, but if I'm not mistaken that function is about the depth initialization of immature point, which is similar to lsd-slam (without the regularization, since we have sparse points in DSO). You can refer to https://jsturm.de/publications/data/engel2013iccv.pdf for an explanation of the depth estimation from repeated epilpolar line search. It also includes a derivation of the error propagation scheme.

More concretely, from the names I'd guess that idepth_min/max refer to the current depth estimate (maybe 3-sigma bounds for a gaussian with mean inverse depth `(idepth_min + idepth_max)/2`.

tanjunyao7, 2019-02-16:
@NikolausDemmel Hi thanks for your explanation. I have investigated the code and the paper a lot these days. It seems that all the magical things start with the variable errorInPixel which quantifies the error of the searched disparity depending on the transformation and the real one on the epipolar line. According to the paper, this variance should be calculated using the formular 6, e.g. the variance of disparity is the variance of epipolar line position over the squared inner product of epi line direction and image gradient direction. However the code is 
`float a = (Vec2f(dx,dy).transpose() * gradH * Vec2f(dx,dy));`
`float b = (Vec2f(dy,-dx).transpose() * gradH * Vec2f(dy,-dx));`
`float errorInPixel = 0.2f + 0.2f * (a+b) / a;`.
It's clear to me that a is the sum of squared inner product of epi line direction and pixel gradient direction in the pattern, e.g. a=Σ(<g,l>)^2 and b the sum of squared inner product of a direction orthogonal to epipolar line and pixel gradient direction, e.g. b=Σ(<g,-1/l>)^2.  But I don't understand where do the 0.2s come from and what (a+b)/a represents for, although it's intuitively straightforward that if g and l are parallel (a big, b=0) the disparity error is small and otherwise large.

vonzy, 2019-03-08:
I am also confused about this uncertainty formula, but it seems everything is ok without it.