JakobEngel/dso #182: Has anyone implemented vi-dso successfully?

vonzy, 2019-03-30:
For the past several months, I have spared no effect to implement vi-dso, but I found it's too hard.

The biggest problem I meet is the rank deficiency of system matrix, since monocular visual-inertial odometry has 4DOF, which means the yaw and the absolute position are unobservable. But in the paper of vidso, the author estimate extra gravity and scale explicitly online with G-N method unlike other tightly-coupled approaches. When I try to implement this approach in vi-dso, the hessian I get is ill-conditioned, it seems imposible to get a good solution. I wonder how to handle ill system matrix to get accurate and stable results.

vonzy, 2019-04-01:
Problem solved

closed 2019-04-01 by vonzy

Tilgaard, 2019-04-02:
Hey Vonzy
Could you please share your solution? Would be nice if you could share your code.

vonzy, 2019-04-02:
@Tilgaard I am still trying to improve the performance of the estimator, if everything is ok, I'll release the code.

robbel, 2020-06-15:
Hi @vonzy did you ever manage to solve the issues and still consider releasing the associated code? Cheers

lukasvst, 2022-05-11:
A (late) update on this: An official implementation for the successor of VI-DSO is now available at: https://github.com/lukasvst/dm-vio