JakobEngel/dso #184: About trajectory estimation

GentlemanCheng, 2019-05-10:
Hi, @Yelen719 .  this work is very nice, I would like to study further, but there is a problem which confused me that how to save the trajectory as a file and compare it with groundtruth by other tools. I have no experience on this, could you give me some suggestions? I will be very grateful if there is corresponding code. Looking forward to your reply. Thank you very much !

GentlemanCheng, 2019-05-10:
@NikolausDemmel Hi, this work is very nice, I would like to study further, but there is a problem which confused me that how to save the trajectory as a file and compare it with groundtruth by other tools. I have no experience on this, could you give me some suggestions? I will be very grateful if there is corresponding code. Looking forward to your reply. Thank you very much !

NikolausDemmel, 2019-05-10:
cross-post: https://github.com/tum-vision/LDSO/issues/26

closed 2019-05-10 by NikolausDemmel