JakobEngel/dso #186: No timestamp in result when mode=1 or mode=2

sanderisbestok, 2019-06-05:
I try to run DSO on the https://daniilidis-group.github.io/penncosyvio/ dataset. It works but when I look at the result I don't see a timestamp, every pose has timestamp 0.

I renamed the files to nanosecond timestamp format but the timestamp will still be zero for all. When I use this mode on the TUM VI dataset with mode 1 or mode 2 I get the same result, no timestamp. 

Any ideas how I can fix this? Or is it a bug?

DRAhmadFaraz, 2019-10-15:
@sanderisbestok can you able to extract poses of every frame image in results.txt file.? 
if yes then please show me your changes steps.

Regards

whut-lxy, 2020-04-17:
> I try to run DSO on the https://daniilidis-group.github.io/penncosyvio/ dataset. It works but when I look at the result I don't see a timestamp, every pose has timestamp 0.
>
> I renamed the files to [...]

Did you solve it? I also encountered this problem when testing the kitti dataset. I hope the first column shows the id of the frame. I  don’t know how to solve it.

StevePhan101, 2020-08-04:
Hi, I have the same problem with all zero timestamps on TUM-VI dataset, sequence 11.
I run with the parameters: mode=0, preset=0.

Cany anyone give me some help?
Thank you.

moajouz, 2024-07-12:
@sanderisbestok fixed?