JakobEngel/dso #187: About Fusion DSO and IMU for Absolute Scale Estimation

studennis911988, 2019-06-12:
Hi, everyone, I'm currently working on fusion visual odometry(DSO) with imu to get absolute scale estimation, and I had read the issue #105  and #36.
The procedure I'm looking for is loosely-coupled filter-based method, and the well-established framework [msf](https://github.com/ethz-asl/ethzasl_msf) turned out to be overwhelming for my system, since I only need the position and scale estimation. 
Therefore I found the paper 
[Fusion of IMU and Vision for Absolute Scale Estimation
in Monocular SLAM](https://www.ifi.uzh.ch/dam/jcr:e885ca94-b971-4bcb-be00-c04b67bebfaa/UAV10_nuetzi.pdf), which algorithm looks much easy and realizable.

I had get the DSO data and implement the paper algorithm by MATLAB, but even how hard I tuned the parameters of each states( position, velocity, acceleration, scale) in processing  covariance and measurement covariance, and It just can't work properly.

-  raw DSO data(red line) and ground truth(black dash line) in same axis
[image]

You can see clearly an initial offset cause by initialization on monocular slam and a scale difference between ground truth(in metric) and visual odometry(without scale)

-  After I reduce the offset manually
[image]

You can see the scale is roughly around 0.5 ( DSO * 0.5 = ground truth)

-  If I fuse DSO(black dash line) and IMU without scale, the result(red line) looks fine without fine tuned parameters
[image]

- The best result (tuning by Genetic Algorithm) including scale in state and apply the proposed algorithm in  [Fusion of IMU and Vision for Absolute Scale Estimation
in Monocular SLAM](https://www.ifi.uzh.ch/dam/jcr:e885ca94-b971-4bcb-be00-c04b67bebfaa/UAV10_nuetzi.pdf)
p.s DSO data (blue line) , fusion data(orange line)

[image]

My guess about the failure is the scale estimate value is wrong, even although I had give the initial   guess close to 0.5, which is the roughly true scale.

My question is 

1.  Has anyone work successfully on this algorithm ? Or any suggestion on this method?
2.  Has anyone try another loosely-coupled filter-based method for visual-inertia fusion?
3.  Moreover, ANY comment is welcomed!
Thanks in advanced!

studennis911988, 2019-06-13:
problem solved !

closed 2019-06-13 by studennis911988

jack-scott, 2019-06-19:
How did you fix this problem?

studennis911988, 2019-06-19:
I put the "unscale" part in measurement and tune by GA, and the results look fine.
[image]