JakobEngel/dso #189: dso_ros

xiaozhi12345678, 2019-07-18:
Hello,I ask the question in the dso_ros, but you haven't answered me,so I try to ask here.I catkin make successfully,but it run 2 seconds then core dumped. I have no ideas about this.

s3r637, 2019-07-22:
Hello @xiaozhi12345678,

excuse me, but what exactly is your question? Do you wanna know how to debug an executable (or a ROS node)?

Cheers.

EdwinvanEmmerik, 2021-03-09:
I think he means that the DSO_ROS package isn't functioning as (at least I) expected with my internal webcam. I spawn the webcam node using usb_cam and pass the correct topic to dso_live with the calibration files.

Problems:
1. Pangolin is demonstrating everything in 2D
2. Core is dumped very frequently

@NikolausDemmel If you want to work this out please contact me and we could work on a fix together. From now I am not sure I understand correctly if dso_ros is actually doing what it is supposed to be doing (no bad intentions ofcourse) but it seems that the implementation is done the right way but what is actually happening is just the image demonstration in Pangolin only.

Best regards,