JakobEngel/dso #191: how to get these files for our own custom images.?

DRAhmadFaraz, 2019-07-27:
@NikolausDemmel Sir, I want to run this SDO to extract the odimetery on my own mono RGB images.? Can you please guide me how to extract the DSO on mono images.?

how to get these parametric files for our own custom images.?

`calib=XXXXX/sequence_XX/camera.txt \`
`gamma=XXXXX/sequence_XX/pcalib.txt \`
`vignette=XXXXX/sequence_XX/vignette.png \`

I will be thankful to you.?

NikolausDemmel, 2019-07-27:
Sorry, I can't guide you step by step. You will need to calibrate your camera yourself. Please see the readme for pointers on how to do that.

qaisarehman, 2019-11-07:
> 
> 
> @NikolausDemmel Thanx for your reply, Can you please guide me how to extract poses of every frame.?

Hi Ahmad! did you manage to figure out how to build your own data set!! Im also stuck on this stage! :)

chapa17, 2021-10-03:
Hello,

Did anyone manage to build their own dataset?