JakobEngel/dso #194: Scaramuzza's camera model?

wendaoyuchen, 2019-08-22:
Hi,thanks for your work! Dose DSO support Scaramuzza's camera model?

https://sites.google.com/site/scarabotix/ocamcalib-toolbox

I calibrate my fisheye camera(fov:210 degree) with PTAM calibrator but RMS cannot be smaller than 0.3, the best result I get is 0.58.
Can you give me some advice? I tried MATLAB 2018a and the fisheye model is Scaramuzza's camera model. And I tried Kalibr with --model omni-none and pinhole-equi, but DSO can't run well with these two models.

NikolausDemmel, 2019-08-22:
The best model supported by DSO for such a wide-angle lense is probably kannala-brandt. You can try basalt to calibrate your lens (https://gitlab.com/VladyslavUsenko/basalt/blob/master/doc/Calibration.md, the model name is "kb4" in basalt).

However, keep in mind that DSO will undistort the images to Pinhole, so it cannot take advantage of the full FOV. So an alternative is also to use some external tools to undistort the images, and then feed those to DSO (with Pinhole model). Maybe the Scaramuzza toolbox can do that.

To take full advantage of the FoV of your camera, you'd have to implement something like described in this paper: https://vision.in.tum.de/_media/spezial/bib/matsuki2018_omnidso.pdf

That version of DSO is not open source, unfortunately.

wendaoyuchen, 2019-08-22:
@NikolausDemmel 
Thanks for your reply! I will try the methods you mentioned. 
In fact, the reason why I try DSO with fisheye camera is that the version of omniDSO is not open source. What I want to do next is implementing omniDSO based on your groups' paper. 
Hope I will achieve it as soon as possible.

QMonteCarlo, 2023-10-09:
you can try DSO-PAL