JakobEngel/dso #199: Error while building " dso "

Shivani1796, 2019-09-03:
/home/shivani/dso/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp: In function ‘dso::MinimalImageB* dso::IOWrap::readImageBW_8U(std::__cxx11::string)’:
/home/shivani/dso/src/IOWrapper/OpenCV/ImageRW_OpenCV.cpp:37:35: error: ‘CV_LOAD_IMAGE_GRAYSCALE’ was not declared in this scope
  cv::Mat m = cv::imread(filename, CV_LOAD_IMAGE_GRAYSCALE);

unable to understand can anyone help .
thanks

ibrahimLearning, 2020-05-27:
@Shivani1796 
have you solved this issue? I have the same problem

jdriscoll319, 2020-10-07:
You need to edit the file to have
`#include <opencv2/imgcodecs.hpp>`
and replace all of the `CV_LOAD_IMAGE_[x]` with `cv::IMREAD_[x]`
e.g. `CV_LOAD_IMAGE_GRAYSCALE` -> `cv::IMREAD_GRAYSCALE`
`CV_LOAD_IMAGE_[x]` is from an older version of openCV and has been deprecated

aswinpajayan, 2021-03-15:
I have both opencv versions installed, is there a way to point to opencv4?

aswinpajayan, 2021-03-15:
> I have both opencv versions installed, is there a way to point to opencv4?

Please ignore, It seems like DSO is using opencv3. If you have multiple versions of opencv installed in your system just point to the right version in the CMakeLists.txt file . 

for me it was in line 23 ;
find_package(OpenCV 3 QUIET)

FishinC, 2022-08-18:
> You need to edit the file to have `#include <opencv2/imgcodecs.hpp>` and replace all of the `CV_LOAD_IMAGE_[x]` with `cv::IMREAD_[x]` e.g. `CV_LOAD_IMAGE_GRAYSCALE` -> `cv::IMREAD_GRAYSCALE` `CV_LOAD_IMAGE_[x]` is from an older version of openCV and has been deprecated

And do not forget to cmake it again: