JakobEngel/dso #201: UndistortFOV::distortCoordinates function

wendaoyuchen, 2019-09-11:
Hi, I am a little confused about the UndistortFOV::distortCoordinates function.
I guess this camera model is Mei model? I cann't understand the following code:
`float d2t = 2.0f * tan(dist / 2.0f);`
`float fac = (r==0 || dist==0) ? 1 : atanf(r * d2t)/(dist*r);`
it's different from Unified Omnidirectional Model in OmniDSO. 
Can you give me any advice or where is the formula above from?

NikolausDemmel, 2019-09-11:
FOV camera model is different from Unified Omnidirectional camera model.

See for a concise description that includes both models: https://arxiv.org/abs/1807.08957

wendaoyuchen, 2019-09-11:
@NikolausDemmel ，thank you !

NikolausDemmel, 2019-09-13:
Please refer to the referenced publication and its references on FOV.

The file you post looks like it may be the OpenCV "fisheye" model. This should correspond to Kannala Brandt with 4 radial parameters.

fangxu622, 2019-09-14:
thanks , i'm begineer ,fishyeye model, I found this
1 open fishyey modelsm, it don't get omega value
2 kalibr , It have fov camera models for get omega ,but it's used on ros and need to make rosbag data.

could post your relative calibration tools or git address .

fangxu622, 2019-09-14:
it 's confusion for me how to make dso data  with our fisheye camera. could post some relative camera calibration tools (url)