JakobEngel/dso #202: Using DSO on TUM-RGBD

ryanpike96, 2019-09-18:
I am trying to compare the VO systems of ORB-SLAM, DSO and SVO and I think TUM-RGBD is a good sequence. Is there a way to run DSO in monocular mode just on the images and timestamps?

Cheers

NikolausDemmel, 2019-09-20:
Yes, thats what is done for Kitti or Euroc for example in ldso, which has some support for these dataset formats, see https://github.com/tum-vision/LDSO/tree/master/examples

You can also convert any sequence to the DSO format and run it with the code from the original DSO repo (use mode=1 if you don't have photometric calibration / exposure times).

However, TUM-RGBD has rolling shutter camera, so DSO will probably not work very well. Check the original paper, where geometric-errors / distortions are discussed. There is also a paper on modelling rolling shutter for DSO (but no code online): https://vision.in.tum.de/_media/spezial/bib/schubert2018drso.pdf

ryanpike96, 2019-09-21:
Thankyou for the response Nikolaus! I have managed to get the files loaded and what not, I was just wondering what the FR1.txt file would be, I.e. the camera calibration? Do I use the RadTan or Pinhole model?

ryanpike96, 2019-09-21:
Nevermind I solved it. Thankyou again.

closed 2019-09-21 by ryanpike96

Rushaonan, 2019-12-27:
I have a same problem, can you tell me how to solve it?

ryanpike96, 2019-12-27:
Unless you model rolling shutter explicitly, DSO will break down pretty quickly on all but the first scene. I tried a fair bit of tuning too. I am not on my computer right now but when I get home I will send you the FR1.txt file that worked for me.

Ryan

Rushaonan, 2019-12-27:
Thank you very much!
I wrote the fr1.txt by imitating the Kitti00-02.txt but it is failed. I don't know how to write images size after calibrated (fr1.txt line 4) and "calib.txt'.What did you do it?

pisuto, 2020-02-02:
@ryanpike96 
hello, I now can run the dataset.
but the config file is wrong.
could you please share the fr1.txt? Thank you very much.