JakobEngel/dso #203: binary `dso_dataset` doesn't build even though OpenCV and Pangolin are both found

rowlanddeep, 2019-09-20:
```
--- found PANGOLIN, compiling dso_pangolin library.
--- found OpenCV, compiling dso_opencv library.
--- found ziplib (1.1.1), compiling with zip capability.
--- compiling dso_dataset.
-- Configuring done
-- Generating done
-- Build files have been written to: /dso/build
...
[100%] Linking CXX executable bin/dso_dataset
[100%] Built target dso_dataset
```
There is no executable binary after it is finished building

chapa17, 2021-09-30:
Hello, I am trying to build dso_dataset but in my case I get a warning that Pangolin could not be found. Can you please let me know whether you installed Pangolin within the dso directory or any other?

Thank you!

IldarGreat, 2022-11-03:
@chapa17 did you solve it?

zerohuwan, 2023-01-07:
> @chapa17 did you solve it?

hi，did you solve it?

IldarGreat, 2023-01-07:
> 
> 
> 
> 
> > @chapa17 did you solve it?
> 
> hi，did you solve it?

Yes, it's need Pangolin v0.6 Try Readme from https://github.com/IldarGreat/dso