JakobEngel/dso #208: how to make every frame as a keyframe and exclude all poses in results.txt file

DRAhmadFaraz, 2019-10-17:
@NikolausDemmel @AlexandrGraschenkov @israelshirk @JakobEngel @Yelen719

how to make every frame as a keyframe and exclude all poses in results.txt file

israelshirk, 2019-10-17:
So keyframes are by definition not every frame, they aggregate data from previous frames in order to reduce the dataset’s complexity for future calculations.  That said, you can get each and every calculated frame (which may be corrected later if I’m remembering which project this is) by implementing your own SampleOutputWrapper as described in this section of the readme:

https://github.com/JakobEngel/dso/blob/master/README.md#34-accessing-data

Copying and pasting from the standard output wrapper into your own logger then tweaking as needed would do the trick.

DRAhmadFaraz, 2019-10-17:
@israelshirk Thanx a lot for replying me and helping me with this issue.

Yes, I have read this readme link and tried to do these `OutputWrapper` but I cant successfully achieve the poses of every frame. 

If you have done this before, can you please send me your `OutputWrapper` file so that I will replace that file on my computer and get the poses of every frame.

I will be thankful to you.

Kallista, 2019-11-19:
could you try if setting setting_onlyLogKFPoses to false will give you poses for all frames?

https://github.com/JakobEngel/dso/blob/e59db70710ead30d6b47959f886665ddd9cb4f85/src/util/settings.cpp#L192