JakobEngel/dso #212: why EAlpha not updated?

rancheng, 2020-01-02:
Recently, I carefully reviewed your code and found the following code very confusing, could you please explain why you accumulate `E` rather than `EAlpha`:

https://github.com/JakobEngel/dso/blob/ae1d0b3d972367b0a384e587852bedcf5f414e69/src/FullSystem/CoarseInitializer.cpp#L502-L519

During the accumulation of energy from depth estimation, you looped all points and just updated `E` in the loop, and later on use the unchanged `EAlpha` to calculate `alphaEnergy`.

rancheng, 2020-01-02:
Another question is that: is this `alphaOpt` set to prevent the energy being too small, since this depends on the translation scale, if the translation is too small, you scale up the error manually by `22500` times (`AlphaW`), and where is this number came from, is it a empirical value?

studennis911988, 2020-01-05:
There is truly some bugs in current code, especially  in initializer part.